vortigont/espem #2: Can not import database with phpMyAdmin

langtupt, 2017-08-31:
Dear Vortigont,
Could you please help me check the database ,i import **pzmysqlstat.sql** in phpmyadmin and do not working.
Please help see the picture below can you help me fix?
Thank you very much
http://prntscr.com/gffkte

vortigont, 2017-09-14:
@langtupt 
You should import only [pzem_mysql.sql](https://github.com/vortigont/espem/blob/master/sql/pzem_mysql.sql) file. it contains all needed db structs. 
**pzmysqlstat.sql** is only reference queries, not used directly anywhere.

langtupt, 2017-09-15:
Dear Vortigont,
Please see the picture problem below. i import pzem_mysql.sql and do not working.
http://prntscr.com/gldelt
please advice and help me fix.

Thank you very much

vortigont, 2017-09-15:
I'm not sure what are you trying to do. Your screen shows that you are already attached to DB 'huuxuanc'.
But 'pzem_mysql.sql' tries to create new DB 'pzem', that's why you get this permission error. If you are not allowed to create new DB's - than remove 'CREATE DATABASE...' and "USE `pzem`;" statements from 'pzem_mysql.sql' and create only tables within your current DB.

IgorVinokur, 2018-02-04:
Dear Vortigont,
Where I can find detailed instruction of the configuration? 

Thanks in Advance

vortigont, 2018-02-04:
@IgorVinokur so sorry, I was to lazy to write the complete guide. But if you have any questions, pls ask, I'll try to help you.

smah1980, 2018-07-26:
@vortigont hello, I am not able to get the view of index.php, I have installed cacti as well as smarty template engine. also set the db parameters in the config.php. Still getting 500 error :(
could you please help where I am doing wrong.

vortigont, 2018-07-26:
Hello @smah1980, pls check your webserver's error log file, it must show what caused the error.
You may also set ini_set('display_errors', 1); in the config.php. But error 500 usually means that cgi execution failed at the early stage, for e.x. missing php module, file access error or something like this.

smah1980, 2018-07-27:
Thanks @vortigont its working now, it was permission issue, I just fixed that, now will check with the meter by tomorrow.
Thanks alot (Y)

vortigont, 2019-01-08:
@smah1980 sorry, was out of office for a while :) just in case if you still heed help on this?

closed 2021-04-13 by vortigont