vortigont/espem #11: How to Download the Code?

Alaa-Aim, 2021-12-19:
Hi Sir
I'm still new on this field, may I know how to download the code to the ESP32? which tool to use for download? is it working on PZEM V3.0?

vortigont, 2021-12-19:
Hi @Alaa-Aim, the project could be build with PlatformIO, you can find more detailed description at the [BUILD](https://github.com/vortigont/espem/blob/master/espem/BUILD.md) page. You also need to upload an FS image to the ESP, there is a shell script for this if you run linux, for windows you might need to install bash for windows, check some online [tutorials](https://www.youtube.com/results?search_query=vscode+platformio+windows+10) for this.
Yes, it works pretty fine with PZEM v3.0

Alaa-Aim, 2021-12-20:
Hi @vortigont , thanks for your prompt response, I'm using Windows, I'll install VS code and platformIO as Add-one, for the FS image, do you mean the SPIFF files ?

vortigont, 2021-12-20:
yeah, only this project uses newer LittleFS instead of SPIFFS. Current Arduino ESP32 framework can't create such an images by default, so you need to download an mklittlefs.exe tool from [this](https://github.com/earlephilhower/mklittlefs/releases) page and drop it to the project's folder where the platformio.ini is and you'll be able to do `pio run -t uploadfs` then.
Do not forget to populate ./data/ dir first by running resources/respack.sh script.

Alaa-Aim, 2021-12-21:
@vortigont thank you for your response, I tried ton build this using VSCode with no success, I always get error in the included libraries, I've tried to install the required libraries, libraries installed successfully but have the red error line under the libraries, any advice ? thank you in advanced

vortigont, 2021-12-21:
@Alaa-Aim 
you do not have to install any libs, platformio should do it during building. Might be something with your setup, maybe 'git' is missing in your system or 'python'? What is your OS - linux or windows? Can you show the error you have in the build terminal console?
I can build you the firmware binaries, not a problem, but it would be tricky to upload  it intro ESP32 without properly working Platformio environment.

Alaa-Aim, 2021-12-21:
many thanks for your response, my OS is Windows, I've installed Arduino, VSCode then added PlatformIO IDE, but I still have the same errors, I installed fresh OS and install Arduino and VSCode only, still the same issue, terminal as below :
[image]

`Processing espem (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.4.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing git+https://github.com/charno/FTPClientServer
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\ALAA\.platformio\platforms\espressif32\builder\main.py", line 233:
    target_elf = env.BuildProgram()
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1104:
    project.install_dependencies()
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 940:
    lm.install(spec)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 93:
    return super(LibraryPackageManager, self)._install(
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141:
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 54:
    obj = getattr(sys.modules[__name__], "%sClient" % type_.title())(
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 138:
    super(GitClient, self).__init__(*args, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 72:
    self.check_client()
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 161:
    raise UserSideException(
================================================== [FAILED] Took 1.07 seconds ==================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:01.075
============================================ 1 failed, 0 succeeded in 00:00:01.075 ============================================ 
The terminal process "C:\Users\ALAA\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.`

Alaa-Aim, 2021-12-21:
Dear @vortigont , already installed git , this is what I got:

[image]
'Processing espem (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.4.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FTP> 0.0.0+sha.f3521ff
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <EmbUI> 2.6.0+sha.cd4ea6d
|   |-- <ArduinoJson> 6.18.5
|   |-- <AsyncMqttClient> 0.9.0
|   |   |-- <AsyncTCP> 1.1.1
|   |-- <TaskScheduler> 3.6.0
|   |-- <LinkedList> 1.4.0+sha.c19a817
|   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <AsyncTCP> 1.1.1
|   |-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |   |-- <AsyncTCP> 1.1.1
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson> 6.18.5
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|   |-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |   |-- <FS> 1.0
|   |-- <Update> 1.0
|-- <pzem-edl> 0.0.4+sha.d8b2a48
|   |-- <LinkedList> 1.4.0+sha.c19a817
|-- <AsyncTCP> 1.1.1
|-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |-- <FS> 1.0
|-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.5
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "D:\ESP\flags.py", line 6, in <module>
    subprocess.check_output(["git", "describe", "--abbrev=4", "--always", "--tags", "--long"])
  File "C:\Users\ALAA\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\ALAA\.platformio\python3\lib\subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', 'describe', '--abbrev=4', '--always', '--tags', '--long']' returned non-zero exit status 128.
OSError: 'python flags.py' exited 1:
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\ALAA\.platformio\platforms\espressif32\builder\main.py", line 233:
    target_elf = env.BuildProgram()
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 159:
    projenv.ProcessFlags(env.get("SRC_BUILD_FLAGS"))
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 223:
    env.Append(**env.ParseFlagsExtended(flags))
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 186:
    for key, value in env.ParseFlags(str(raw)).items():
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Environment.py", line 815:
    do_parse(arg)
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Environment.py", line 667:
    arg = self.backtick(arg[1:])
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Environment.py", line 579:
    raise OSError("'%s' exited %d" % (command, status))
================================================== [FAILED] Took 2.23 seconds ==================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:02.226
============================================ 1 failed, 0 succeeded in 00:00:02.226 ============================================ 
The terminal process "C:\Users\ALAA\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
'

vortigont, 2021-12-21:
ugh...
`subprocess.CalledProcessError: Command '['git', 'describe', '--abbrev=4', '--always', '--tags', '--long']' returned non-zero exit `
looks like that's a Windows issue... pretty ugly OS :)
I'll fix that and let you know, need to find some windows hosts to test :))))

vortigont, 2021-12-21:
Meanwhile you can workaround it,
open 'platformio.ini' file and replace this line `!python flags.py` with `-DGIT_REV="git_xxx"`
Mind the spaces indent!

Alaa-Aim, 2021-12-21:
@vortigont many thanks for your replied and patient, really appreciate it, wait to hear from you, bear with me, I'm still noob in this field and still learning.

vortigont, 2021-12-21:
@Alaa-Aim you are wecome!
You may also face some issues with building FS image under windows. Actually never tested it, because it requires some unix tools to be available under windows, usually comes with git-bash installed.
Anyway, here is an FS image file for you attached. Just drop it under .pio/build/espem
[spiffs.zip](https://github.com/vortigont/espem/files/7756303/spiffs.zip)

Alaa-Aim, 2021-12-21:
@vortigont I did replaced '!python flags.py' with '-DGIT_REV="git_2.34.1"', still have error, thanks for the SPIFFS

[image]

'Processing espem (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.4.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FTP> 0.0.0+sha.f3521ff
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <EmbUI> 2.6.0+sha.cd4ea6d
|   |-- <ArduinoJson> 6.18.5
|   |-- <AsyncMqttClient> 0.9.0
|   |   |-- <AsyncTCP> 1.1.1
|   |-- <TaskScheduler> 3.6.0
|   |-- <LinkedList> 1.4.0+sha.c19a817
|   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <AsyncTCP> 1.1.1
|   |-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |   |-- <AsyncTCP> 1.1.1
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson> 6.18.5
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|   |-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |   |-- <FS> 1.0
|   |-- <Update> 1.0
|-- <pzem-edl> 0.0.4+sha.d8b2a48
|   |-- <LinkedList> 1.4.0+sha.c19a817
|-- <AsyncTCP> 1.1.1
|-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |-- <FS> 1.0
|-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.5
Building in release mode
Replace MKSPIFFSTOOL with mklittlefs.exe
Compiling .pio\build\espem\src\espem.cpp.o
Compiling .pio\build\espem\src\interface.cpp.o
Compiling .pio\build\espem\src\main.cpp.o
Compiling .pio\build\espem\lib0ad\FS\vfs_api.cpp.o
<command-line>:0:14: error: too many decimal points in number
espem\main.cpp:106:5: note: in expansion of macro 'GIT_REV'
     GIT_REV,
     ^
Archiving .pio\build\espem\lib0ad\libFS.a
espem\main.cpp: In function 'void wver(AsyncWebServerRequest*)':
<command-line>:0:9: error: 'git_2' was not declared in this scope
espem\main.cpp:106:5: note: in expansion of macro 'GIT_REV'
     GIT_REV,
     ^
Compiling .pio\build\espem\lib5f0\WiFi\ETH.cpp.o
*** [.pio\build\espem\src\main.cpp.o] Error 1
================================================== [FAILED] Took 7.47 seconds ==================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:07.471
============================================ 1 failed, 0 succeeded in 00:00:07.471 ============================================ 
The terminal process "C:\Users\ALAA\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
'

vortigont, 2021-12-21:
OK, here is the real revision string `-DGIT_REV='"v2.0.1-10-gfba6"'`, try this one

vortigont, 2021-12-21:
or maybe omit single quotes, just -DGIT_REV="v2.0.1-10-gfba6"
not sure how windoze likes it :)

Alaa-Aim, 2021-12-21:
> or maybe omit single quotes, just -DGIT_REV="v2.0.1-10-gfba6" not sure how windose likes it :)

Windows like it with two quotes, compile is successful but when connect to it and navigate to 192.168.4.1 it shows "not found", I remembered to upload FS, when I ran `platformio run -t uploadfs` , this is what i got:

PS D:\ESP> platformio run -t uploadfs
Can not remove temporary directory `D:\ESP\.pio\build`. Please remove it manually to avoid build issues
Processing espem (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.4.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FTP> 0.0.0+sha.f3521ff
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <EmbUI> 2.6.0+sha.cd4ea6d
|   |-- <ArduinoJson> 6.18.5
|   |-- <AsyncMqttClient> 0.9.0
|   |   |-- <AsyncTCP> 1.1.1
|   |-- <TaskScheduler> 3.6.0
|   |-- <LinkedList> 1.4.0+sha.c19a817
|   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <AsyncTCP> 1.1.1
|   |-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |   |-- <AsyncTCP> 1.1.1
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson> 6.18.5
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|   |-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |   |-- <FS> 1.0
|   |-- <Update> 1.0
|-- <pzem-edl> 0.0.4+sha.d8b2a48
|   |-- <LinkedList> 1.4.0+sha.c19a817
|-- <AsyncTCP> 1.1.1
|-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |-- <FS> 1.0
|-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.5
Building in release mode
Replace MKSPIFFSTOOL with mklittlefs.exe
Building SPIFFS image from 'data' directory to .pio\build\espem\spiffs.bin
'"./mklittlefs.exe"' is not recognized as an internal or external command,
operable program or batch file.
*** [.pio\build\espem\spiffs.bin] Error 1
================================================== [FAILED] Took 1.83 seconds ==================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:01.831
============================================ 1 failed, 0 succeeded in 00:00:01.831 ============================================

Alaa-Aim, 2021-12-21:
`PS D:\ESP> platformio run -t upload
Can not remove temporary directory `D:\ESP\.pio\build`. Please remove it manually to avoid build issues
Processing espem (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.4.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FTP> 0.0.0+sha.f3521ff
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <EmbUI> 2.6.0+sha.cd4ea6d
|   |-- <ArduinoJson> 6.18.5
|   |-- <AsyncMqttClient> 0.9.0
|   |   |-- <AsyncTCP> 1.1.1
|   |-- <TaskScheduler> 3.6.0
|   |-- <LinkedList> 1.4.0+sha.c19a817
|   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <AsyncTCP> 1.1.1
|   |-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |   |-- <AsyncTCP> 1.1.1
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson> 6.18.5
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|   |-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |   |-- <FS> 1.0
|   |-- <Update> 1.0
|-- <pzem-edl> 0.0.4+sha.d8b2a48
|   |-- <LinkedList> 1.4.0+sha.c19a817
|-- <AsyncTCP> 1.1.1
|-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |-- <FS> 1.0
|-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.5
Building in release mode
Replace MKSPIFFSTOOL with mklittlefs.exe
Retrieving maximum program size .pio\build\espem\firmware.elf
Checking size .pio\build\espem\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.9% (used 42200 bytes from 327680 bytes)
Flash: [========  ]  78.8% (used 1033181 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\espem\firmware.bin
esptool.py v3.1
Serial port COM3
Connecting........_
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:61:05:2e:a6:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0010cfff...
Compressed 17104 bytes to 11191...
Writing at 0x00001000... (100 %)
Wrote 17104 bytes (11191 compressed) at 0x00001000 in 0.6 seconds (effective 231.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 306.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 500.4 kbit/s)...
Hash of data verified.
Compressed 1033296 bytes to 589295...
Writing at 0x00010000... (2 %)
Writing at 0x0001b479... (5 %)
Writing at 0x00029ca8... (8 %)
Writing at 0x000328cf... (11 %)
Writing at 0x00049785... (13 %)
Writing at 0x00052ea1... (16 %)
Writing at 0x00058fa8... (19 %)
Writing at 0x0005f24f... (22 %)
Writing at 0x000650c5... (25 %)
Writing at 0x0006a23f... (27 %)
Writing at 0x000711dd... (30 %)
Writing at 0x00076f20... (33 %)
Writing at 0x0007e3e9... (36 %)
Writing at 0x0008836d... (38 %)
Writing at 0x0008dd85... (41 %)
Writing at 0x00093d28... (44 %)
Writing at 0x000996c8... (47 %)
Writing at 0x0009f7ee... (50 %)
Writing at 0x000a5580... (52 %)
Writing at 0x000ab3cc... (55 %)
Writing at 0x000b16ab... (58 %)
Writing at 0x000b7a94... (61 %)
Writing at 0x000bd84c... (63 %)
Writing at 0x000c3a0d... (66 %)
Writing at 0x000c8b1c... (69 %)
Writing at 0x000cdf48... (72 %)
Writing at 0x000d33c0... (75 %)
Writing at 0x000d899d... (77 %)
Writing at 0x000ddd92... (80 %)
Writing at 0x000e3443... (83 %)
Writing at 0x000e8d88... (86 %)
Writing at 0x000ef06d... (88 %)
Writing at 0x000f4306... (91 %)
Writing at 0x000fa717... (94 %)
Writing at 0x001008b5... (97 %)
Writing at 0x00106529... (100 %)
Wrote 1033296 bytes (589295 compressed) at 0x00010000 in 14.8 seconds (effective 557.4 kbit/s)...

Leaving...
Hard resetting via RTS pin...
================================================= [SUCCESS] Took 27.99 seconds =================================================
Environment    Status    Duration
-------------  --------  ------------
espem          SUCCESS   00:00:27.991
================================================= 1 succeeded in 00:00:27.991 ================================================= `

vortigont, 2021-12-21:
Nice! :)))
Check [BUILD](https://github.com/vortigont/espem/blob/master/espem/BUILD.md) page, you need mklittlefs,exe binary to make FS iamges.

> l but when connect to it and navigate to 192.168.4.1 it shows "not found"
you can also navigate to http://192.168.4.1/update and upload fs image I gave you from there

Alaa-Aim, 2021-12-21:
> Nice! :))) Check [BUILD](https://github.com/vortigont/espem/blob/master/espem/BUILD.md) page, you need mklittlefs,exe binary to make FS iamges.
>
>
> you can also navigate to http://192.168.4.1/update [...]

I navaigate to 192.168.4.1/update, and uploaded the SPIFFS you gave me, the user interface worked but I can't change the name of AP and I can't connect to any WIFI network, any easy way for me to compile my own SPIFFS? really appreciate you response and replies.

vortigont, 2021-12-21:
No need, you can do all from WebUI.
Internal AP name is hostname actually, can change under the 'device name' section.
WiFI connection is set under WiFi Client section, need to fill both SSID and password, there is no scanning for available AP's around. Once connected, esp will shutdown internal AP and you can access it via http://hostname/ in your network.

Alaa-Aim, 2021-12-21:
> 

Actually I did all you mentioned, but when I enter the SSID & Password (entered manually as no scanning functionality) , ESP rest but nothing changed, still in AP mode and I can't change the host name nor connect to the WiFi , all above things I did it from webUI , but no success

Alaa-Aim, 2021-12-21:
[image]

vortigont, 2021-12-21:
Ugh... not sure. Pls, try to build your own FS image as described in DOCS. Or you can try to clear config from "System setup" page.
Anyway you can try to connect your PZEM device right away and check it :)))
Going offline for today, will be back tomorrow.
Cheers!

Alaa-Aim, 2021-12-21:
> device right away and check it :)))

I managed to make my FS 
`alaa85@DESKTOP-KSFF4NQ:/mnt/d/ESP/resources$ ./respack.sh
Preparing resources for ESPEM FS image into ../data/ dir
EmbUI resources requires updating
Refreshing EmbUI css files/pics...
./respack.sh: 112: unzip: not found
refreshing embui js files...
alaa85@DESKTOP-KSFF4NQ:/mnt/d/ESP/resources$`

this the files I had , 
[image]

still have the same error when I run `platformio run -t uploadfs`

"PS D:\ESP> platformio run -t uploadfs
Can not remove temporary directory `D:\ESP\.pio\build`. Please remove it manually to avoid build issues
Processing espem (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.4.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FTP> 0.0.0+sha.f3521ff
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <EmbUI> 2.6.0+sha.cd4ea6d
|   |-- <ArduinoJson> 6.18.5
|   |-- <AsyncMqttClient> 0.9.0
|   |   |-- <AsyncTCP> 1.1.1
|   |-- <TaskScheduler> 3.6.0
|   |-- <LinkedList> 1.4.0+sha.c19a817
|   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <AsyncTCP> 1.1.1
|   |-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |   |-- <AsyncTCP> 1.1.1
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson> 6.18.5
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|   |-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |   |-- <FS> 1.0
|   |-- <Update> 1.0
|-- <pzem-edl> 0.0.4+sha.d8b2a48
|   |-- <LinkedList> 1.4.0+sha.c19a817
|-- <AsyncTCP> 1.1.1
|-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |-- <FS> 1.0
|-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.5
Building in release mode
Replace MKSPIFFSTOOL with mklittlefs.exe
Building SPIFFS image from 'data' directory to .pio\build\espem\spiffs.bin
'"./mklittlefs.exe"' is not recognized as an internal or external command,
operable program or batch file.
*** [.pio\build\espem\spiffs.bin] Error 1
=========================================================== [FAILED] Took 1.88 seconds ===========================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:01.878
===================================================== 1 failed, 0 succeeded in 00:00:01.878 ====================================================="

Alaa-Aim, 2021-12-21:
@vortigont I managed to upload the SPIFFS I made, the languge become Russian (I guess) and I still have the same issue, can't change the device name, can't connect any WiFi even though I've tried many WiFis with no luck

vortigont, 2021-12-21:
@Alaa-Aim OK, I was able to run this under windoze :) 
I was even able to run without -DGIT_REV hack. I use windows 7, not sure if Win10 has some tricks, heard from other ppl that it's better to install python via Microsoft store or whatever it is called... nevermind. Actually you can just simply remove this line from platformio.ini, should be fine.
I've made some small fixes in repo, you need to redownload it from scratch, rebuild firmware and upload it into MCU,. Hope this will fix WiFi issue, even might not need to rebuild FS image. Just checked with a brand new esp32 board - it works pretty fine :)

What I did is installed git for windows with that feature enabled
[image]

than I was able build fw and run respack.sh script right from the VSCode terminal
[image]

> I managed to make my FS ` ./respack.sh: 112: unzip: not found refreshing embui js files... alaa85@DESKTOP-KSFF4NQ:/mnt/d/ESP/resources$`
> 
> this the files I had ,

That is not right! You miss unzip, so you actually have nothing. the directory with data is not `resources`, it's just sources. The real data for LittleFS goes into `./data` dir. You should have there at least `index.html.gz` and css/js dirs.

> * Building SPIFFS image from 'data' directory to .pio\build\espem\spiffs.bin
>       '"./mklittlefs.exe"' is not recognized as an internal or external command,

you are still missing `mklittlefs.exe`. AGAIN, pls read [BUILD.md](https://github.com/vortigont/espem/blob/master/espem/BUILD.md) and download binary mklittlefs.exe for windows from specified [REPO](https://github.com/earlephilhower/mklittlefs/releases) and place it into the repo's dir.

> the languge become Russian (I guess)

you can change it back to English from "settings" page - top drop down list.

Alaa-Aim, 2021-12-22:
@vortigont , thanks for your effort and response, I downloaded the new respo, compiling was not success with the included platformIO.ini, i had to use the one we made yesterday, and compiling was successful, FS build was successful to after I placed the mklittlefs.exe in resources folder and I got index.htmland css/js dirs.
`alaa85@DESKTOP-KSFF4NQ:/mnt/d/ESP/resources$ ./respack.sh
Preparing resources for ESPEM FS image into ../data/ dir
EmbUI is up to date
refreshing embui js files...
alaa85@DESKTOP-KSFF4NQ:/mnt/d/ESP/resources$`
the I used the two commands to upload the code and the FS
`platformio run -t upload`
`platformio run -t uploadfs`

everything was successful, but when I connect to the ESP for configuration, the image below is what I got, I used brand new MCU and I Still have this issue 

[image]

vortigont, 2021-12-22:
> everything was successful, but when I connect to the ESP for configuration, the image below is what I got

this is still wrong, you are missing css/js files in the ./data/ dir, this means respack.sh script failed to download new files.
And firmware source files are wrong - EmbUI version must be 2.6.1 with new sources. Try to delete project folder completely and download again from GitHub

Alaa-Aim, 2021-12-22:
@vortigont I removed the whole folder and downloaded a new one, the FS is made with no errors, below screen shots for the files i had, 
[image]
[image]
[image]
[image]

when i use the provided .ini file, I get error when compiling as shown below :
"Processing espem (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.4.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FTP> 0.0.0+sha.f3521ff
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <EmbUI> 2.6.0+sha.cb88d7f
|   |-- <ArduinoJson> 6.18.5
|   |-- <AsyncMqttClient> 0.9.0
|   |   |-- <AsyncTCP> 1.1.1
|   |-- <TaskScheduler> 3.6.0
|   |-- <LinkedList> 1.4.0+sha.c19a817
|   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <AsyncTCP> 1.1.1
|   |-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |   |-- <AsyncTCP> 1.1.1
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson> 6.18.5
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|   |-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |   |-- <FS> 1.0
|   |-- <Update> 1.0
|-- <pzem-edl> 0.0.4+sha.d8b2a48
|   |-- <LinkedList> 1.4.0+sha.c19a817
|-- <AsyncTCP> 1.1.1
|-- <LittleFS_esp32> 1.0.6+sha.f0817ca
|   |-- <FS> 1.0
|-- <ESP Async WebServer> 1.2.3+sha.fdd1b43
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.5
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "D:\ESP32\flags.py", line 6, in <module>
    subprocess.check_output(["git", "describe", "--abbrev=4", "--always", "--tags", "--long"])
  File "C:\Users\ALAA\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\ALAA\.platformio\python3\lib\subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', 'describe', '--abbrev=4', '--always', '--tags', '--long']' returned non-zero exit status 128.
OSError: 'python flags.py' exited 1:
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\ALAA\.platformio\platforms\espressif32\builder\main.py", line 233:
    target_elf = env.BuildProgram()
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 159:
    projenv.ProcessFlags(env.get("SRC_BUILD_FLAGS"))
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 223:
    env.Append(**env.ParseFlagsExtended(flags))
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ALAA\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 186:
    for key, value in env.ParseFlags(str(raw)).items():
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Environment.py", line 815:
    do_parse(arg)
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Environment.py", line 667:
    arg = self.backtick(arg[1:])
  File "C:\Users\ALAA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Environment.py", line 579:
    raise OSError("'%s' exited %d" % (command, status))
================================================== [FAILED] Took 2.08 seconds ==================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:02.083
============================================ 1 failed, 0 succeeded in 00:00:02.083 ============================================ 
The terminal process "C:\Users\ALAA\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1."

and when I use the file we modified yesterday it compiles with no errors.

and still get the same page as last comment.

please advise, thanks for your efforts and response

vortigont, 2021-12-22:
> below screen shots for the files i had,

somehow you miss css files, you need to check the output of respack.sh script, maybe it misses some tool.
Anyway, here is the data dir content, just unpack it to yours ./data/
[data.zip](https://github.com/vortigont/espem/files/7762469/data.zip)

> and when I use the file we modified yesterday it compiles with no errors.

do not use yesterday's platformio.ini file. Use new one, just remove the line `!python flags.py`
Your Windows is nasty :)

vortigont, 2021-12-22:
Always clean `.pio` folder on new rebuild, it may contain previous versions of libs

Alaa-Aim, 2021-12-22:
> Always clean `.pio` folder on new rebuild, it may contain previous versions of libs

@vortigont may you teach me to clean. pio folder? many thanks

vortigont, 2021-12-22:
just delete it :) it's under project's folder

Alaa-Aim, 2021-12-22:
> just delete it :) it's under project's folder

I did that, I deleted the whole folder and copy new respo. and I still have the same issue

vortigont, 2021-12-22:
did you removed `!python flags.py` line from `platformio.ini` file? Does the build succeeded?

Alaa-Aim, 2021-12-22:
@vortigont after many tries, it connected to the WiFi and I can access it from the WebUI, but I can't change the device name the web interface same not right, below screen shots 

[image]
[image]
[image]

vortigont, 2021-12-22:
> web interface same not right

your FFS image misses .css files, same as before. Unpack the [archive](https://github.com/vortigont/espem/files/7762469/data.zip) to your ./data/ dir and reflash FS.

Or out can access internal FS via FTP client and upload missing files directly to the MCU

Alaa-Aim, 2021-12-22:
@vortigont Now it works fine after uploading the new `,/data`  folder, may I know how to fix my FS files? how to build files same as what you uploaded?
 below screen shots for the new webUI and for the files i had inside my `resources` folder, many thanks for your response and efforts

[image]
[image]
[image]
[image]
[image]

vortigont, 2021-12-22:
> may I know how to fix my FS files?

Not sure, you need to check the output of the respack.sh script, it might throw some error why it can't unpack css files. You can run it via git-bash shell terminal. Also it might be useful to run it with `-f` flag to force resources update.
`mklittlefs,exe` must be placed to the project's root dir, right near the `platformio.ini` file
Are you able to change the hostname now?

Alaa-Aim, 2021-12-22:
> Not sure, you need to check the output of the respack.sh script, it might throw some error why it can't unpack css files. You can run it via git-bash shell terminal. Also it might be useful to run [...]

I placed the mklittlefs.exe in the root directory and when I run '. /respack.sh' no error appears, but when I cross checked the files, it seems less files generated.

host name, I still not able to change, I'll do what you mentioned and revert back to you.

vortigont, 2021-12-22:
> now all files generated correctly, upload it to the ESP32 and works fine, many thanks for your prompt responses and efforts.

great! Finally :) u r welcome )

Alaa-Aim, 2021-12-26:
Dear @vortigont , Merry Christmas , sorry for bothering you again, I wanted to modified you project a little, I wanted the ESP32 to trigger an action when the voltage goes below or above a threshold (i.e. turn on a LED), but couldn't find the WebUI getting the voltage value from, cloud you please point this out for me, so I can use to trigger the "action"

[image]

vortigont, 2021-12-26:
@Alaa-Aim, Better open new issue next time, not to mix topics.
UI update task is triggered in method [ESPEM::wspublish()](https://github.com/vortigont/espem/blob/cf2daac20c2610e45a0a762c1e8ac5bfa5a87875/espem/espem.cpp#L220) 
You can hack into there for testing only, but it's not the best place due to it does not run when no WebUI clients are present `if (!embui.ws.count()`

Or you can hijack to the call-back [lambda](https://github.com/vortigont/espem/blob/cf2daac20c2610e45a0a762c1e8ac5bfa5a87875/espem/espem.cpp#L285-L291
) of the sampler function, it runs every second, collecting sampled data regardless of the WebUI clients.

But much better way would be to create your own function like `ESPEM::wspublish()` and attach it as a separate scheduled task to run periodically
https://github.com/vortigont/espem/blob/cf2daac20c2610e45a0a762c1e8ac5bfa5a87875/espem/espem.cpp#L59-L61

Aleks130699, 2022-01-12:
Good afternoon, I also had problems compiling on windows 10, but like deleting a line !python flags.py from platformio.ini, helped in compilation, there were no problems with fs compilation, connected pzem 004t v3 to esp8266, receives information, but not how I can't force to connect to wifi, while if wifi is without a password, then esp connects without problems.

vortigont, 2022-01-12:
@Aleks130699 
ok, first question - you are working with esp8266 or esp32? Current master branch does not support esp8266.
Regarding WiFi - what is your issue? Are you able to access WebUI via internal AP and enter WiFi credentials for your WiFi-AP?

vortigont, 2022-01-12:
> like deleting a line !python flags.py from platformio.ini, helped in compilation
tnx for confirmation, I do not have win10 to test this. Could you show a console output or a screenshot with an error.
I'm already thinking about completely disabling this script for Windows :)

Aleks130699, 2022-01-12:
I use esp8266, everything works through an internal access point, I can enter data for wifi, but esp does not connect to my wifi, and creates an access point again, but if you remove the security from my wifi, then esp works fine through my network

Aleks130699, 2022-01-12:
Все работает только не подключается к wifi....
[image]

vortigont, 2022-01-12:
@Aleks130699 попробуйте собрать версию с дебагом, в build_flags нужно добавить
```
  -DEMBUI_DEBUG
  -DEMBUI_DEBUG_PORT=Serial
```
или   -DEMBUI_DEBUG_PORT=Serial1  если первый порт занят
вероятно это проблема совместимости 8266 и роутеров от асус, бага висит на гитхабе.
Как вариант еще можене попробовать и/или добавить флаг  `-DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190313`
это более старый сдк, но с ним 8266 нормально работает в режиме 802.11n

vortigont, 2022-01-12:
ну если дебаг сможете получить, то было бы хорошо. На 8266 это сложно, там второй сериал порт оч неудобно сделан. 
 Переходите на esp32 :) там совсем другая библиотека используется.
У меня часы работают на 8266, вроде к сети подключаются, но в последнее время не тестировал подробно.
Лампа это форк 100 летней давности, я работал над тем проектом когда-то. По ней лучше в оригиналный проект смотреть.

Aleks130699, 2022-01-12:
А возможности нет скомпилировать для дальнейший загрузки через node mcu flasher?, а то нет линукса

Aleks130699, 2022-01-12:
Ну да старая версия, ну вообще по сути она же только одна подходит под esp8266 и pzem 004t v3 ?

vortigont, 2022-01-12:
да. так вам как собрать? У вас как пзем подключен, на 0-й порт и родные пины? Я использую `ESPEM_HWSERIAL_SWAP` и дебаг на 1й порт, это получается GPIO2

Aleks130699, 2022-01-12:
Да вообще без разницы, у меня сейчас на 1 и 3 пин подключен

vortigont, 2022-01-12:
собрал со старым сдк 2 версии - с дебагом и без, плюс образ ФС. Залить можно через вебюи, через флешер я сходу не помню какие там адреса надо выставлять.
Дебаг на гпио2, пзем на пинах по умолчанию
[espem_fw.zip](https://github.com/vortigont/espem/files/7856626/espem_fw.zip)

Aleks130699, 2022-01-12:
Не, чет все равно не хочет, хотя веб интерфейс по другому стал работать, анимации появились, а версия с дебагом не загружает есп, даже точки ап не появляется

vortigont, 2022-01-12:
хех... придется найти 8266 пустую и потестить... там в новом core 3.x много чего понаменяли, может где-то и выстрелило.
Дам знать позже

Aleks130699, 2022-01-12:
Главное блин если сеть буз пароля, то подключается и работает

Aleks130699, 2022-01-12:
И mqtt походу тоже не але....Нужно в общем заказывать esp32...

vortigont, 2022-01-13:
да, проблему с коннектом воспроизвел. Оч странно причем, на часах с этим же фреймворком все работает. Буду разбираться.

> И mqtt походу тоже не але...

mqtt у меня в прошивке не реализован, как-то нужды не было. Настройки в вебюи это заготовка из фреймворка, но она там по сути пустая. Весь функционал надо самому писать. Если будет желание заняться тестированием и разработкой схемы в помощь, могу заняться этим чуть позже. В любом случае придется рано или поздно :)

Aleks130699, 2022-01-13:
есп32 уже заказал, буду ждать, а если у вас нет mqtt то как забираете данные куда то дальше? или так только в вебуи смотрите?

[25 more replies]

closed 2023-12-14 by vortigont