vortigont/espem #12: Getting Values From PEZM

Alaa-Aim, 2021-12-27:
@vortigont , Sorry for the inconvenient for putting the new issue with the old issue, your explaining in previous replay , I didn't fully get it as I am still new in programing, as far as I know that the PZEM uses the library `pzem-edl` to get the data from PZEM module, with the library there is an example and tried it and I got the values I want , but when I combined it with the Main.cpp in the espem folder, the ESP32 is crashes, please advise.
Thanks in advanced.

vortigont, 2021-12-27:
> but when I combined it with the Main.cpp in the espem folder, the ESP32 is crashes, please advise.

yes, in main.cpp you already have `pzem-edl`s pzem object available
try something like this in `main.cpp`

```
void mycheck();
Task t_checker;

void setup(){
  //run check every seconds
  t_checker.set( TASK_SECOND, TASK_FOREVER, mycheck );
  ts.addTask(t_checker);
  t_checker.enableDelayed();
}

void mycheck(){
  const auto m = espem->pz->getMetricsPZ004();
  if (m->voltage/10 > 250){
    Serial.println("High voltage!");
    //digitalwrite(alert_LED, HIGH);
  }
}

```

vortigont, 2021-12-27:
> P.S: the WebUI does not work on mobile phone, any work around ?

it works pretty good on mobile, via embedded browser, I have no issues.
I use [Service Browser](https://play.google.com/store/apps/details?id=com.druk.servicebrowser) app to discover EmbUI devices in the local network, so you do not even need to enter it's URL, just open WWW HTTP service

Alaa-Aim, 2021-12-27:
> yes, in main.cpp you already have `pzem-edl`s pzem object available try something like this in `main.cpp`
>
> ```
> void mycheck();
> Task t_checker;
>
> void setup(){
>   //run check every seconds [...]

@vortigont thanks for your prompt response, the triggering action works fine but nothing printed on the serial monitor, `void mycheck()` runs every second, right? I just wanted to check the interval

Alaa-Aim, 2021-12-27:
> it works pretty good on mobile, via embedded browser, I have no issues. I use [Service Browser](https://play.google.com/store/apps/details?id=com.druk.servicebrowser) app to discover EmbUI devices [...]

@vortigont I used the Service Browser and it works fine, many thanks.

closed 2021-12-27 by Alaa-Aim