vortigont/espem #15: Reading Stucks After Main Disconncted From PEZEM

Alaa-Aim, 2022-02-05:
@vortigont , after playing around with your project, I faced that the reading stuck on the last reading i.e. 221 Volt after disconnect the mains (main voltage supply to PZEM i.e. 220V or 240V)from PZEM module, without resetting the ESP32.
as example, if I disconnected the mains from PZEM module while keep the ESP32 running, the readings from PZEM, voltage, frequency, power and energy, stuck at the last reading not going back to zero as expected. as when the mains from PZEM disconnected all the readings shall be come zero as no input fed to the PZEM module.
Please advise.

vortigont, 2022-02-05:
Indeed, it's true. But it is "by design" in a new lib. You can have "zeroes" only if you really got 0 volts reply from PZEM, but with disconnected mains it means "no reply" at all. Actually the lib provides "age" time for the last valid read. I.e. 221 volt, 2 amps, etc, + age 600000. It means that last valid reply was received 10 minutes ago.
This behavior is not adopted for the espem WebUI yet. Not sure how properly handle it - just simply display zeroes via WebIU or return some NaNs. I will think about it.

Alaa-Aim, 2022-02-06:
@vortigont many thanks , waiting for your reply, I will try to find a way around.

vortigont, 2022-02-06:
What is your issue? If you do not like WebUI showing old data, than fix is pretty easy, I'll post one tomorrow.
If you interact with lib API than you can make a simple age check same way as for TimeSeries data. 
https://github.com/vortigont/espem/blob/59fb94b82b897306062ff3b6bfe8f5a99320f707/espem/espem.cpp#L287-L290

Alaa-Aim, 2022-02-06:
> What is your issue? If you do not like WebUI showing old data, than fix is pretty easy, I'll post one tomorrow. If you interact with lib API than you can make a simple age check same way as for [...]

@vortigont my issue is , I want to check if the main for PZEM module goes off or disconnected, I want to trigger an action, i.e., switching alarm or change source.

vortigont, 2022-02-06:
Than it's not an issue at all, use something like this

```
if (pz->getState()->dataStale()){ 
   Serial.printf("Connection to PZEM has been lost %lld seconds ago\n", pz->getState()->dataAge() / 1000 );
 } 
```

Alaa-Aim, 2022-02-06:
> 

espem/main.cpp:285:73: error: 'pz' was not declared in this scope
@vortigont I got this error

Alaa-Aim, 2022-02-06:
@vortigont but i got the error below 
[E][vfs_api.cpp:64] open(): /littlefs/index.html does not exist
Connection to PZEM has been lost 0 seconds ago
[E][vfs_api.cpp:64] open(): /littlefs/samples.json.gz does not exist
[E][vfs_api.cpp:64] open(): /littlefs/samples.json does not exist
[E][vfs_api.cpp:64] open(): /littlefs/samples.json/index.html.gz does not exist
[E][vfs_api.cpp:64] open(): /littlefs/samples.json/index.html does not exist

Alaa-Aim, 2022-02-06:
> Than it's not an issue at all, use something like this
> 
> ```
> if (pz->getState()->dataStale()){ 
>    Serial.printf("Connection to PZEM has been lost %lld seconds ago\n", pz->getState()->dataAge() / 1000 );
>  } 
> ```

this code to be in espem/espem/espem.cpp or in the main.cpp

vortigont, 2022-02-06:
I do not mean that you can blindly put those lines into random place and it will do the magic for you. It's just a method to check last time a proper response has been received from PZEM. It's up to you to decide how often do those checks and where to put the code. If you put it into loop() it will do the checks 10000 times per second - not the best way obviously.
pz is the member of EPSEM class. If you need to access it in main you should do it via espem object, i.e. `espem->pz->getState()->dataAge()` will return you last access time in milliseconds, 0-1200 ms under normal conditions. It's up to you to decide what to do with this info.

vortigont, 2022-02-06:
Check lib example for similar case
https://github.com/vortigont/pzem-edl/blob/9750b664b68908340d9679b8b6ee936e260e693c/examples/01_SinglePZEM004/src/main.cpp#L122-L132

Alaa-Aim, 2022-02-07:
> I do not mean that you can blindly put those lines into random place and it will do the magic for you. It's just a method to check last time a proper response has been received from PZEM. It's up to you to decide how often do those checks and where to put the code. If you put it into loop() it will do the checks 10000 times per second - not the best way obviously. pz is the member of EPSEM class. If you need to access it in main you should do it via espem object, i.e. `espem->pz->getState()->dataAge()` will return you last success time in milliseconds, 0-1200 ms under normal conditions. It's up to you to decide what to do with this info.

@vortigont I made a separate loop for it with task scheduler and it works , many hanks

commit 720251dc (2022-02-07): UI can detect if PZEM is disconnected or unreachable and show "err" values

+ some minor formatting fixes for UI displays

Closes #15
closed 2022-02-07 by Alaa-Aim