vortigont/espem #16: TX & RX in WebUI

Alaa-Aim, 2022-02-10:
@vortigont after the last updates for all libraries in the espem project, I faced an issue with the TX & RX in the WebUI, if the internet connection disconnected or the ESP32 restarted, the connection between PZEM and the ESP32 will be lost, I have to go to the WebUI, ESPEM Setup, then hit Apply for the RX & TX Pins, then the connection restored, I thought it is an ESP32 module issue, I changed 3 modules with the same issue.

vortigont, 2022-02-10:
Hi @Alaa-Aim 
thanks for continuing testing my firmware :)
I can't reproduce the issue, it works fine for me. Maybe you config file become corrupted? Pls, try to access "Settings" -> "system settings" -> "clear sys config" and then "reboot"
System config file is available via url http://espem/config.json
It should contain "rx":18,"tx":22 keys. 18/22 is just an example. Default config contains "-1"
Also you need to update FS for the latest changes I've made to js display's code.

Alaa-Aim, 2022-02-10:
@vortigont I did cleared the config file and reboot, the issue persistent, the pines i use are 16 & 17 these values are not changed, I have to access the WebUI and change nothing only hit apply, then everything works fine again, I recompiled the firmware and regenerated the FS, cleaned the .pio folder, I still get the same error, I will try to used ESP32 from a different manufacture, right now I use AI Thinker

vortigont, 2022-02-10:
Are you sure you run my version without modifications?
Looks like this lines is missins in your setup
https://github.com/vortigont/espem/blob/720251dc1fd77adae908498da9803e6b2e3a53c5/espem/main.cpp#L49-L54

Otherwise, pls build "espem_debug" version and check the serial log output on boot.
```
UI WiFi: STA reconecting...
espem.begin: port: 1, rx_pin: 18, tx_pin:22
Autopolling enabled
SRAM: heap 348976, free 233204
SPI-RAM: heap 0, free 0

```