vortigont/espem #17: All Loops On Hold Whene WiFi Disconnected

Alaa-Aim, 2022-02-18:
@vortigont , Many thanks for your replies and effort for this project, I'm using it to monitor some inputs and outputs, when WiFi disconnected, the ESP puts everything on hold and only AutoConnect works, by other mean, all loops will not working only the Autoconnect and nothing works till the connection restored, this feature / property, interrupts what I monitor and all inputs & outputs reading will be on hold, I want to make AutoConnect works when WiFi connection lost but all the loops keep running, is this possible? please bear with me, and many thanks again.

vortigont, 2022-02-19:
Hi!
What do you mean loops on hold? The Arduino loop() function? Actually pzem-edl lib works out of loop() cycle in a separate RTOS task, so it should not be affected. As for the other calls from loop() all of those are called one by one and can be blocked if some of the tasks locks the loop. WiFi could block for air scanning, etc... If you want to run some tasks asynchronously than you need to use interrupt calls, queues, thread core binding and other RTOS features. Concurrent thread-safe programing is not that easy but doable. There is no nice and easy solution here, but you can start from the official Espressif [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#semaphore-api)

Alaa-Aim, 2022-02-20:
@vortigont Thanks for your reply, what I meant is all voids and task i.e. void loop, void mycheck, void check, etc. will be will be on hold and not working when the ESP32 lost WiFi connection , even the threshold trigging will not work when the WiFi connection lost.
I thought the Autoconnect is part of your code and I want disable the blocking, to let the other voids working while the Autoconnect activated.

vortigont, 2022-02-20:
Yes, WiFi is managed via EmbUI framework. But it does not stop loop() completely. There might be some deviations in timings, but still works.
Test this
```
uint32_t t = 0;
// MAIN loop
void loop() {

  if (millis() - t > 1000){
    t = millis();
    LOG(printf, "Time: %d", t);
  }

  embui.handle();

#ifdef USE_FTP
    ftp_loop(); // цикл обработки событий фтп-сервера
#endif
}
```

then disconnect WiFi, and you can see this in debug log

```
Time: 59059ws[/ws][1] disconnect
Time: 60060Time: 61061Time: 62062UI WiFi: Disconnected, reason: 200
UI WiFi: Reconnect attempt
Time: 63063Time: 64064UI WiFi: Disconnected, reason: 201
Time: 65065Time: 66066UI WiFi: Disconnected, reason: 201
Time: 67067Time: 68068UI WiFi: Disconnected, reason: 201
Time: 69069Time: 70070UI WiFi: Disconnected, reason: 201
Time: 71071Time: 72072UI WiFi: Switch to AP/STA mode
E (73052) wifi:sta is connecting, return error
[E][WiFiSTA.cpp:221] begin(): connect failed!
UI WiFi: Disconnected, reason: 201
Time: 73073Time: 74074Time: 75075UI WiFi: Disconnected, reason: 201
Time: 76076Time: 77077UI WiFi: Disconnected, reason: 201
Time: 78078Time: 79079Time: 80080UI WiFi: Disconnected, reason: 201
Time: 81081Time: 82082UI WiFi: Disconnected, reason: 201
Time: 83083Time: 84084Time: 85085UI WiFi: Disconnected, reason: 201
Time: 86086Time: 87087UI WiFi: Disconnected, reason: 201
Time: 88088Time: 89089Time: 90090UI WiFi: Disconnected, reason: 201
```

BTW, if do not need WiFi you can enable "AP-only" mode in WiFi settings. In this mode esp will switch to it's internal Aceess Point and will never make attempts to reconnect to external WiFi effectively disabling WiFi client.

Alaa-Aim, 2022-02-20:
@vortigont , I added the lines you mentioned above, I got the below response
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
E (34470) wifi:sta is connecting, return error
[E][WiFiSTA.cpp:221] begin(): connect failed!`

I noticed the the hold for the voids is about 2 to 3 seconds to change from station to access point, for some voids it's ok but some voids the timing is critical.

Can I make the ESP32 to work as access point only in the initial setup ? as I can change the WiFi settings from webUI once configured.

vortigont, 2022-02-20:
That means that some of YOUR functions or tasks are blocking the loop(). Without seeing the whole project I can tell you more.
Suppose the one that tries to access the network while WiFi is disconnected. Revise it and find the one that blocks.

> Can I make the ESP32 to work as access point only in the initial setup ? as I can change the WiFi settings from webUI once configured.

I do not understand how this will prevent you from disconnections, but you can switch to AP-only mode in the WiFi settings.

Alaa-Aim, 2022-02-20:
@vortigont , Many thanks, I will remove the functions required accessing the WiFi and see what will happen, thanks again

vortigont, 2022-02-20:
you are welcome!

Alaa-Aim, 2022-02-22:
@vortigont I removed all functions connection to WiFi and I still have the same, once the WiFi disconnected, most of the functions looks like resting but the ESP32 not resetting, I thing I will make it as AP, as you advise to prevent such thing from happening. thanks again.

closed 2022-02-22 by Alaa-Aim