vortigont/espem #18: Use ESP32 WROVER with 4MB PSRAM

Alaa-Aim, 2022-03-01:
@vortigont I want to use this project on ESP32 WROVER with 16 MB Flash memory & 4 MB PSRAM, how to utilize the PSRAM for this project ?  usage of 16 MB memory how can be done?  and what changes in the .ini file needs to be done? Thanks in advanced

vortigont, 2022-03-03:
Hi @Alaa-Aim 
it should work out of the box. PSRAM could be used to allocate circular buffers for time series data.
So you would be able to increase it's size considerably and collect sampling history for much longer time. This feature is still needs further improvement, like samling daily/hourly/minute stat. I'll implement it once I find some free time slot for this :)

No special settings required, you just need to specify your board name in platformio.ini's env. Once PSRAM is detected [pzem-edl](https://github.com/vortigont/pzem-edl) will use it to store samples.
If build with debuging enabled you could see a message in console log like
```
"SRAM: heap %d, free %d"
"SPI-RAM: heap %d, free %d\n"
```

For the platformio.ini you may need something like this, although I do not remember if those keys are still mandatory.
```
[env:esp32psram]
platform = espressif32
board = esp-wrover-kit
framework = arduino
build_flags =
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
```

Pls, provide your feedback one tested :) It's much appreciated!

Alaa-Aim, 2022-03-04:
@vortigont , I have ESP32 DevKitC V4 , WROVER with 16MB Memory & 4MB PSRAM, I modified the platformio.ini file as you mentioned in you reply, below what showed up in serial monitor :

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Starting EspEM...
UI: Config file load OK!
UI: Creating system vars
UI: Creating application vars
UI REGISTER: b_espem
UI REGISTER: b_emset
UI: Creating webui vars
UI REGISTER: settings
UI REGISTER: sh_sec
UI REGISTER: shname
UI REGISTER: set_wifi
UI REGISTER: set_wifiAP
UI REGISTER: set_mqtt
UI REGISTER: set_time
UI REGISTER: lang
UI REGISTER: reboot
UI REGISTER: time
UI REGISTER: s_cfgcl
UI REGISTER: a_setem
UI REGISTER: a_uart
UI REGISTER: a_pzopt
UI REGISTER: dctl*
UI READ KEY: 'm_pref' VAL: '818F14'
UI READ KEY: 'm_host' VAL: ''
UI READ KEY: 'm_port' VAL: ''
UI READ KEY: 'm_user' VAL: ''
UI READ KEY: 'm_pass' VAL: ''
UI: MQTT host is empty - disabled!
UI CONFIG: {"m_host":"","m_port":"","m_user":"","m_pass":"","m_pref":"818F14","m_tupd":"30","dctlupdrt":2,"ems_prd":1,"emplsz":300,"uart":1,"rx":-1,"tx":-1,"eoffset":0}UI READ KEY: 'TZSET' VAL: ''
UI WiFi: start in STA mode
[E][WiFiSTA.cpp:221] begin(): connect failed!
generate autohostname: EmbUI-818F14
UI WiFi: STA reconecting...
espem.begin: port: 1, rx_pin: -1, tx_pin:-1
Autopolling enabled
SRAM: heap 321732, free 214344
SPI-RAM: heap 4194236, free 4185836
UI WiFi: set AP params to SSID:EmbUI-818F14, pwd:
UI mDNS: responder started: EmbUI-818F14.local
UI WiFi: Switch to AP/STA mode

woks fine, looking foreword for your update for the sampling, daily, hourly .....etc.

Alaa-Aim, 2022-03-04:
@vortigont after making the env:esp32psram default environment a get the error in compiling as below :

Compiling .pio\build\esp32psram\lib0ad\FS\vfs_api.cpp.o
In file included from espem\espem.cpp:9:0:
espem\espem.h:11:24: fatal error: pzem_edl.hpp: No such file or directory

********************************************************************
* Looking for pzem_edl.hpp dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pzem_edl.hpp"
* Web  > https://registry.platformio.org/search?q=header:pzem_edl.hpp
*
********************************************************************

compilation terminated.
In file included from espem\interface.cpp:1:0:
espem\espem.h:11:24: fatal error: pzem_edl.hpp: No such file or directory

********************************************************************
* Looking for pzem_edl.hpp dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pzem_edl.hpp"
* Web  > https://registry.platformio.org/search?q=header:pzem_edl.hpp
*
********************************************************************

compilation terminated.
*** [.pio\build\esp32psram\src\espem.cpp.o] Error 1
*** [.pio\build\esp32psram\src\interface.cpp.o] Error 1
In file included from espem\main.cpp:11:0:
espem\espem.h:11:24: fatal error: pzem_edl.hpp: No such file or directory

********************************************************************
* Looking for pzem_edl.hpp dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pzem_edl.hpp"
* Web  > https://registry.platformio.org/search?q=header:pzem_edl.hpp
*
********************************************************************

compilation terminated.
*** [.pio\build\esp32psram\src\main.cpp.o] Error 1
============================================================ [FAILED] Took 6.75 seconds ============================================================
Environment    Status    Duration
-------------  --------  ------------
esp32psram     FAILED    00:00:06.746

Alaa-Aim, 2022-03-05:
after playing around, I changed the TX & RX pins in the WwbUI, in EPS32 WROOM Rx pin 3 & Tx pin 1, in ESP32 WORVER Rx pin 1 & Tx Pin 3, but the memory still not changed, seems like the PSRAM not utilized as shown in screen shoot below
please advise .... 
[image]

vortigont, 2022-03-07:
Hi @Alaa-Aim, the WebUI shows only SRAM memory heap. PSRAM heap is not displayed sine not all boards has PSRAM and I do not want to display confusing zeros there. The actual info could be seen in console log as you've posted above.

Alaa-Aim, 2022-03-07:
Hi @vortigont many thanks for your efforts and replies, so the log seems fine? the PSRAM been utilized ? but the memory seems to be lower than noraml ESP32, i mean lower that the ESP31 without PSRAM!! is it OK?

closed 2022-03-07 by vortigont

vortigont, 2022-03-07:
check recently updated code. Should be able to get PSRAM stat via http://espem/fw

Alaa-Aim, 2022-03-07:
Hi @vortigont thanks for your prompt action and reply, screen shoot below what I got 
[image]

[image]

Many thanks in advanced

vortigont, 2022-03-07:
oops, sorry, this is not a valid json format, I will fix it later.
but anyway you can read the values :) first number is 'total', second is 'free'

Alaa-Aim, 2022-03-07:
@vortigont msny thanks for your efforts..

commit 2b1fa098 (2022-03-07): add PSRAM heap size info to /fw responce
add PZEM data printout to debug output

Close #18