vortigont/espem #20: Error While Compiling (After Update)

Alaa-Aim, 2022-05-18:
@vortigont after the update of EmbUI library , I get error while compiling as shown below in compiling log & screen shoot

In file included from espem/espem.h:12,
                 from espem/espem.cpp:9:
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:20:2: error: #error "Sorry, but this lib requires c++14 compatible compiler to build"
 #error "Sorry, but this lib requires c++14 compatible compiler to build"
  ^~~~~
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:21:2: error: #error "check supplied examples on how to add proper build flags for platformio"        
 #error "check supplied examples on how to add proper build flags for platformio"
  ^~~~~
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:22:10: fatal error: no_c++14: No such file or directory
 #include "no_c++14"
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\espem\src\espem.cpp.o] Error 1
In file included from espem/espem.h:12,
                 from espem/interface.cpp:1:
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:20:2: error: #error "Sorry, but this lib requires c++14 compatible compiler to build"
 #error "Sorry, but this lib requires c++14 compatible compiler to build"
  ^~~~~
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:21:2: error: #error "check supplied examples on how to add proper build flags for platformio"        
 #error "check supplied examples on how to add proper build flags for platformio"
  ^~~~~
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:22:10: fatal error: no_c++14: No such file or directory
 #include "no_c++14"
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\espem\src\interface.cpp.o] Error 1
In file included from espem/espem.h:12,
                 from espem/main.cpp:11:
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:20:2: error: #error "Sorry, but this lib requires c++14 compatible compiler to build"
 #error "Sorry, but this lib requires c++14 compatible compiler to build"
  ^~~~~
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:21:2: error: #error "check supplied examples on how to add proper build flags for platformio"        
 #error "check supplied examples on how to add proper build flags for platformio"
  ^~~~~
.pio/libdeps/espem/pzem-edl/src/timeseries.hpp:22:10: fatal error: no_c++14: No such file or directory
 #include "no_c++14"
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\espem\src\main.cpp.o] Error 1
=========================================================== [FAILED] Took 14.83 seconds ===========================================================

Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:14.826

[image]

vortigont, 2022-05-20:
I was able to update to 6.0.1, I use linux though
```
-(0)> pio --version
PlatformIO Core, version 6.0.1
```
dev version might have surprises need to keep an eye on a dev log

Alaa-Aim, 2022-10-27:
@vortigont , after updating EmbUI, when compiling, I get an error, as below:

.pio/libdeps/espem/EmbUI/EmbUI/EmbUI.cpp: In member function 'void EmbUI::section_handle_remove(const String&)':
.pio/libdeps/espem/EmbUI/EmbUI/EmbUI.cpp:228:28: error: 'LList<std::shared_ptr<section_handle_t> >' {aka 'class LL::LinkedList<std::shared_ptr<section_handle_t> >'} 
has no member named 'unlink'
             section_handle.unlink(i);
                            ^~~~~~
*** [.pio\build\espem\libed0\EmbUI\EmbUI.cpp.o] Error 1
==================================================================== [FAILED] Took 19.69 seconds ====================================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:19.695

[image]

Please advise

closed 2022-10-27 by Alaa-Aim