vortigont/espem #21: Error Uploading FS

Alaa-Aim, 2022-05-22:
@vortigont , sorry for my many questions , I changed my PC, installed VSCode and PlatformIO , I compiled the project and upload it to ESP32, it was successful, but when I tried to upload the FS i got an error as show below 👍 
PS D:\ESPEM-WROOM> platformio run -t uploadfs
Processing espem (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Library Manager: Installing Hash
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (4.3.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.0 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FTP @ 0.0.0+sha.f3521ff
|   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- EmbUI @ 2.6.2+sha.4520447
|   |-- ArduinoJson @ 6.19.4
|   |-- AsyncMqttClient @ 0.9.0
|   |   |-- AsyncTCP @ 1.1.1
|   |-- TaskScheduler @ 3.6.0
|   |-- LinkedList @ 1.4.0+sha.0c8c344
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- HTTPClient @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |-- ESP Async WebServer @ 1.2.3+sha.fdd1b43
|   |   |-- AsyncTCP @ 1.1.1
|   |   |-- FS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- ArduinoJson @ 6.19.4
|   |-- FS @ 2.0.0
|   |-- LittleFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- Update @ 2.0.0
|-- pzem-edl @ 1.0.2+sha.c964cab
|   |-- LinkedList @ 1.4.0+sha.0c8c344
|-- Adafruit GFX Library @ 1.11.1
|   |-- Adafruit BusIO @ 1.11.6
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- Adafruit BusIO @ 1.11.6
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- Adafruit SSD1306 @ 2.5.3
|   |-- Adafruit GFX Library @ 1.11.1
|   |   |-- Adafruit BusIO @ 1.11.6
|   |   |   |-- Wire @ 2.0.0
|   |   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- AsyncTCP @ 1.1.1
|-- LittleFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- HTTPClient @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
|-- ESP Async WebServer @ 1.2.3+sha.fdd1b43
|   |-- AsyncTCP @ 1.1.1
|   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- ArduinoJson @ 6.19.4
Building in release mode
Replace MKSPIFFSTOOL with mklittlefs.exe 
*** [C:\Users\ALAA\.platformio\packages\tool-mklittlefs\mklittlefs.EXE] D:\ESPEM-WROOM\.pio\build\espem\.sconsign39.dblite: No such file or directory
==================================================================== [FAILED] Took 4.47 seconds ====================================================================

Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:04.468
=============================================================== 1 failed, 0 succeeded in 00:00:04.468 ===============================================================PS D:\ESPEM-WROOM>

vortigont, 2022-05-23:
Hi @Alaa-Aim 
I do not have Windows env to test it. You can try and comment out
`extra_scripts = replace_fs.py` in platformio.ini [esp32_base] section.
Newer Arduino core should be able to build LittleFS image on it's own I hope.
Then there is no need in extra tool for this.
Let me know if it helps.

Alaa-Aim, 2022-05-23:
@vortigont , I did comment it out and it works, then I comment it out and works again , many thanks for your replies

commit 9c52b70f (2022-05-23): update platformio.ini config for Arduino core 2.0.x

external mklittlefs tool is no longer required

Closes #21
commit c4602c64 (2022-05-23): update platformio.ini config for Arduino core 2.0.x

external mklittlefs tool is no longer required

Closes #21
commit c89bd80d (2022-05-23): update platformio.ini config for Arduino core 2.0.x

external mklittlefs tool is no longer required

Closes #21