vortigont/espem #22: Randam Restart

Alaa-Aim, 2022-05-27:


vortigont, 2022-05-28:
Hello @Alaa-Aim, no problem, any questions are welcome :)
Random restarts are most hard to debug. A new core introduced lots of changes in almost all components so could expect regressions. Most fragile part is async server, it has not been updated long time up to modern changes. Try to collect decoded stack traces and conditions when those crashes could happen. At least to narrow down the areas for possible issues.

Alaa-Aim, 2022-05-28:
@vortigont , when build and upload(both FS and FW) in debug mode, am almost had no restart, but I can't find the device in (Service Browser ), when I access the ESP32 by browser , I had no restart at all , below some of what shown in the debug console 

UI: ws[/ws][3] connect MEM: 191948
UI: section begin:'root111880216' [#0] 7989 free
UI: section begin:'menu' [#1] 7936 free
UI: Frame add obj 68 b, mem:256/8192...OK idx:0	MEM: 183472
UI: Frame add obj 64 b, mem:340/8192...OK idx:1	MEM: 183500
UI: Frame add obj 62 b, mem:408/8192...OK idx:2	MEM: 183500
UI: section end:'menu' [#1] MEM: 183520
UI: json_frame_flush
UI: section end:'root111880216' [#0] MEM: 183584
UI: section begin:'root111917078' [#0] 8080 free
UI: section begin:'b_espem' [#1] 7962 free
UI: section begin:'' [#2] 7892 free
UI: Frame add obj 166 b, mem:300/8192...OK idx:0	MEM: 182956
UI: Frame add obj 233 b, mem:476/8192...OK idx:1	MEM: 182912
UI: section end:'' [#2] MEM: 182976
UI: section begin:'' [#2] 7443 free
UI: Frame add obj 154 b, mem:749/8192...OK idx:0	MEM: 182948
UI: Frame add obj 154 b, mem:894/8192...OK idx:1	MEM: 182948
UI: Frame add obj 156 b, mem:1022/8192...OK idx:2	MEM: 182948
UI: section end:'' [#2] MEM: 182976
UI: section begin:'' [#2] 6976 free
UI: Frame add obj 198 b, mem:1216/8192...OK idx:0	MEM: 182948
UI: Frame add obj 198 b, mem:1392/8192...OK idx:1	MEM: 182948
UI: section end:'' [#2] MEM: 182976
UI: Frame add obj 217 b, mem:1541/8192...OK idx:3	MEM: 183020
UI: Frame add obj 226 b, mem:1709/8192...OK idx:4	MEM: 183020
UI: json_frame_flush
UI: section end:'b_espem' [#1] MEM: 183012
UI: section begin:'root111996648' [#0] 8084 free
UI: Frame add obj 74 b, mem:108/8192...OK idx:0	MEM: 181724
UI: json_frame_flush
UI: section end:'root111996648' [#0] MEM: 181772
UI: section begin:'root112008072' [#0] 916 free
UI: Frame add obj 74 b, mem:108/1024...OK idx:0	MEM: 180316
UI: Frame add obj 67 b, mem:192/1024...OK idx:1	MEM: 180316
UI: Frame add obj 70 b, mem:261/1024...OK idx:2	MEM: 180316
UI: json_frame_flush
UI: section end:'root112008072' [#0] MEM: 180336
UI: section begin:'root112164331' [#0] 402 free
UI: Frame add obj 47 b, mem:110/512...OK idx:0	MEM: 192928
UI: Frame add obj 45 b, mem:173/512...OK idx:1	MEM: 192928
UI: Frame add obj 43 b, mem:225/512...OK idx:2	MEM: 192928
UI: Frame add obj 43 b, mem:275/512...OK idx:3	MEM: 192928
UI: Frame add obj 43 b, mem:325/512...OK idx:4	MEM: 192928
UI: Frame add obj 43 b, mem:375/512...OK idx:5	MEM: 192928
UI: Frame add obj 44 b, mem:425/512...OK idx:6	MEM: 192928
UI: json_frame_flush
UI: section end:'root112164331' [#0] MEM: 192956
[112253][E][vfs_api.cpp:104] open(): /littlefs/samples.json.gz does not exist, no permits for creation
[112255][E][vfs_api.cpp:104] open(): /littlefs/samples.json does not exist, no permits for creation
[112262][E][vfs_api.cpp:104] open(): /littlefs/samples.json/index.html.gz does not exist, no permits for creation
[112272][E][vfs_api.cpp:104] open(): /littlefs/samples.json/index.html does not exist, no permits for creation
TimeSeries buffer has 104 items, scntr: 0
Sending timeseries JSON, buffer 5549/5585, items left: 31
Sending timeseries JSON, buffer 1444/1455, items left: 12

[320369][E][vfs_api.cpp:104] open(): /littlefs/samples.json/index.html does not exist, no permits for creation
TimeSeries buffer has 300 items, scntr: 0
ws[/ws][24] disconnect

vortigont, 2022-05-28:
Tnx for details, ServiceBrowser feature is handled via Arduino's mDNS library, might be the issue there or need to adopt to recent changes. I will try to check this when time permits. Had not tested new framework much lately.
Keep this open for a while.

vortigont, 2022-05-30:
I can confirm an issue with mDNS lib. Somehow ServiceDiscovery tool crashes esp32 almost 100% on it's first run. Not sure I can find a workaround for this. Meanwhile I can suggest not to use ServiceDiscovery or look for some alternatives. Browser access via http://espname/ work pretty fine.

vortigont, 2022-05-30:
Found the bug that causes the issue - espressif/arduino-esp32#6368

Alaa-Aim, 2022-05-30:
@vortigont it seems the mDNS issue with Android 8 and above, only accessing using IP address through normal browser (not Services Browser) the safe method of accessing the WebUI right now!

Alaa-Aim, 2022-05-31:
so using the [https://github.com/vortigont/espem/tree/feat](url) make it easier to access the WebUI? or easier to make a shortcut for WebUI accessing?

vortigont, 2022-05-31:
you need to (at least) upload updated resources from that branch - files `resources/html/manifest.json` and *.png's to the root of your esp32 devise. that access your device with chrome browser on android via http://device/, than use menu-add_desktop shortcut in chrome.

Alaa-Aim, 2022-05-31:
Well noted and many thanks

vortigont, 2022-05-31:
let's keep it open till fixed in Arduino core

Alaa-Aim, 2022-08-31:
@vortigont , after the last updates, I tried the Service Browser with 2 different devices and 3 mobile phones, it seems the issue no longer exist,