vortigont/espem #23: Error Compiling

Alaa-Aim, 2022-07-14:
@vortigont After the last PlatformIO, I have error compiling, as shown in the log below :

Archiving .pio\build\espem\lib4d4\libHTTPClient.a
Compiling .pio\build\espem\liba89\esp32-flashz\flashz.cpp.o
In file included from .pio/libdeps/espem/esp32-flashz/src/flashz-http.cpp:30:
.pio/libdeps/espem/esp32-flashz/src/flashz-http.hpp:36:10: fatal error: WebServer.h: No such file or directory

*******************************************************************
* Looking for WebServer.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WebServer.h"
* Web  > https://registry.platformio.org/search?q=header:WebServer.h
*
*******************************************************************

 #include <WebServer.h>
          ^~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\espem\libab2\LittleFS\LittleFS.cpp.o
*** [.pio\build\espem\liba89\esp32-flashz\flashz-http.cpp.o] Error 1
==================================================================== [FAILED] Took 31.21 seconds ====================================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:31.211

vortigont, 2022-07-14:
Hi @Alaa-Aim,
ops! sorry, didn't want to affect current master. I'm working on compressed OTA updates with [esp32-flashz](https://github.com/vortigont/esp32-flashz). Will fix this shortly.

Alaa-Aim, 2022-07-14:
@vortigont many thanks for prompt reply

Alaa-Aim, 2022-07-15:
@vortigont , another error appears , as the log below 

Building in release mode
Compiling .pio\build\espem\liba89\esp32-flashz\flashz-http.cpp.o
In file included from .pio/libdeps/espem/esp32-flashz/src/flashz-http.cpp:30:
.pio/libdeps/espem/esp32-flashz/src/flashz-http.hpp:36:10: fatal error: WebServer.h: No such file or directory

*******************************************************************
* Looking for WebServer.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WebServer.h"
* Web  > https://registry.platformio.org/search?q=header:WebServer.h
*
*******************************************************************

 #include <WebServer.h>
          ^~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\espem\libe87\EmbUI\http.cpp.o
Compiling .pio\build\espem\libe87\EmbUI\wi-fi.cpp.o
Compiling .pio\build\espem\libe87\EmbUI\wifi_esp32.cpp.o
Compiling .pio\build\espem\libe87\EmbUI\wifi_esp8266.cpp.o
Compiling .pio\build\espem\libaa0\pzem-edl\modbus_crc16.cpp.o
Compiling .pio\build\espem\libaa0\pzem-edl\msgq.cpp.o
Compiling .pio\build\espem\libaa0\pzem-edl\pzem_edl.cpp.o
*** [.pio\build\espem\liba89\esp32-flashz\flashz-http.cpp.o] Error 1
.pio/libdeps/espem/EmbUI/EmbUI/http.cpp:12:10: fatal error: flashz-async.hpp: No such file or directory

************************************************************************
* Looking for flashz-async.hpp dependency? Check our library registry!
*
* CLI  > platformio lib search "header:flashz-async.hpp"
* Web  > https://registry.platformio.org/search?q=header:flashz-async.hpp
*
************************************************************************

 #include "flashz-async.hpp"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\espem\libe87\EmbUI\http.cpp.o] Error 1
==================================================================== [FAILED] Took 36.07 seconds ====================================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:36.074

commit 8a0799bb (2022-07-14): add build flag for esp32-flashz lib

closes #23
closed 2022-07-15 by Alaa-Aim