vortigont/espem #25: SyntaxError: JSON.parse

Alaa-Aim, 2022-09-16:
@vortigont, when trying access "http://espem/fw", I get this error 
"SyntaxError: JSON.parse: end of data when property name was expected at line 1 column 200 of the JSON data"

vortigont, 2022-09-16:
Hello @Alaa-Aim,
unfortunately I do not have a working PZEM at hand now. Can you post here a json reply? Might be some comma or bracket missing.

Alaa-Aim, 2022-09-16:
@vortigont , I've tried to look for this error in the webpage files, no luck, could please point me to the file responsible of this error 

[image]

vortigont, 2022-09-16:
it's a browser's error. Just click on a 'raw' tab to see json body

Alaa-Aim, 2022-09-16:
Tried different browser and no error , thanks

closed 2022-09-16 by Alaa-Aim