vortigont/espem #26: ESP32S3 With 2MB PSRAM, Compiling Error

Alaa-Aim, 2022-10-07:
@vortigont, I have ESP32S3 with 2MB PSRAM, tried to use your code on this board, first time, I compiled it & upload it but nothing worked, no WIFI nor any output, then I realized that I didn't change the board in the *.ini file, I changed it to be like the screenshot below 

[image]

but I get compiling error as below ;

`Compiling .pio\build\espem\lib7c2\EmbUI\wifi_esp32.cpp.o
In file included from C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/HTTP_Method.h:4,
                 from C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:30,
                 from .pio/libdeps/espem/esp32-flashz/src/flashz-http.hpp:36,
                 from .pio/libdeps/espem/EmbUI/EmbUI/http.cpp:12:
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:95:6: error: 'HTTP_DELETE' conflicts with a previous declaration
   XX(0,  DELETE,      DELETE)       \
      ^
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX' #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from .pio/libdeps/espem/EmbUI/EmbUI/EmbUI.h:49,
                 from .pio/libdeps/espem/EmbUI/EmbUI/http.cpp:10:
.pio/libdeps/espem/AsyncWebServerMod/src/ESPAsyncWebServer.h:64:3: note: previous declaration 'WebRequestMethod HTTP_DELETE'
   HTTP_DELETE  = 0b00000100,
   ^~~~~~~~~~~
In file included from C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/HTTP_Method.h:4,
                 from C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:30,
                 from .pio/libdeps/espem/esp32-flashz/src/flashz-http.hpp:36,
                 from .pio/libdeps/espem/EmbUI/EmbUI/http.cpp:12:
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:96:6: error: 'HTTP_GET' conflicts with a 
previous declaration
   XX(1,  GET,         GET)          \
      ^
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX' #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from .pio/libdeps/espem/EmbUI/EmbUI/EmbUI.h:49,
                 from .pio/libdeps/espem/EmbUI/EmbUI/http.cpp:10:
.pio/libdeps/espem/AsyncWebServerMod/src/ESPAsyncWebServer.h:62:3: note: previous declaration 'WebRequestMethod HTTP_GET'
   HTTP_GET     = 0b00000001,
   ^~~~~~~~
In file included from C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/HTTP_Method.h:4,
                 from C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:30,
                 from .pio/libdeps/espem/esp32-flashz/src/flashz-http.hpp:36,
                 from .pio/libdeps/espem/EmbUI/EmbUI/http.cpp:12:
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:97:6: error: 'HTTP_HEAD' conflicts with a previous declaration
   XX(2,  HEAD,        HEAD)         \
      ^
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX' #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from .pio/libdeps/espem/EmbUI/EmbUI/EmbUI.h:49,
                 from .pio/libdeps/espem/EmbUI/EmbUI/http.cpp:10:
.pio/libdeps/espem/AsyncWebServerMod/src/ESPAsyncWebServer.h:67:3: note: previous declaration 'WebRequestMethod HTTP_HEAD'
   HTTP_HEAD    = 0b00100000,
   ^~~~~~~~~
In file included from C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/HTTP_Method.h:4,
                 from C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:30,
                 from .pio/libdeps/espem/esp32-flashz/src/flashz-http.hpp:36,
                 from .pio/libdeps/espem/EmbUI/EmbUI/http.cpp:12:
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:98:6: error: 'HTTP_POST' conflicts with a previous declaration
   XX(3,  POST,        POST)         \
      ^
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX' #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
[... 56 lines cut ...]
   HTTP_PATCH   = 0b00010000,
   ^~~~~~~~~~
.pio/libdeps/espem/EmbUI/EmbUI/http.cpp: In member function 'void EmbUI::http_set_handlers()':
Compiling .pio\build\espem\lib7c2\EmbUI\wifi_esp8266.cpp.o
.pio/libdeps/espem/EmbUI/EmbUI/http.cpp:113:44: error: no matching function for call to 'FlashZhttp::provide_ota_form(AsyncWebServer*, const char*&)'
     fz.provide_ota_form(&server, UPDATE_URI);
                                            ^
In file included from .pio/libdeps/espem/EmbUI/EmbUI/http.cpp:12:
.pio/libdeps/espem/esp32-flashz/src/flashz-http.hpp:164:10: note: candidate: 'void FlashZhttp::provide_ota_form(WebServer*, const char*)'
     void provide_ota_form(WebServer *srv, const char* url);
          ^~~~~~~~~~~~~~~~
.pio/libdeps/espem/esp32-flashz/src/flashz-http.hpp:164:10: note:   no known conversion for argument 1 from 'AsyncWebServer*' to 'WebServer*'
.pio/libdeps/espem/EmbUI/EmbUI/http.cpp:114:43: error: no matching function for call to 'FlashZhttp::handle_ota_form(AsyncWebServer*, const char*&)'
     fz.handle_ota_form(&server, UPDATE_URI);
                                           ^
In file included from .pio/libdeps/espem/EmbUI/EmbUI/http.cpp:12:
.pio/libdeps/espem/esp32-flashz/src/flashz-http.hpp:175:10: note: candidate: 'void FlashZhttp::handle_ota_form(WebServer*, const char*)'
     void handle_ota_form(WebServer *server, const char* url);
          ^~~~~~~~~~~~~~~
.pio/libdeps/espem/esp32-flashz/src/flashz-http.hpp:175:10: note:   no known conversion for argument 1 from 'AsyncWebServer*' to 'WebServer*'
*** [.pio\build\espem\lib7c2\EmbUI\http.cpp.o] Error 1
==================================================================== [FAILED] Took 43.26 seconds ====================================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:43.261`

Alaa-Aim, 2022-10-07:
I removed the build flags, and I still get the same error 

[image]

Alaa-Aim, 2022-10-07:
The error is in the EmbUI Library, as shown in the screenshot below 

[image]

Alaa-Aim, 2022-10-07:
@vortigont , I did the above and it works, I have one issue in the WebUI, as you know the U0RX, U0TX are pin 44 & 43 respectively, when I enter these number the WebUI do not accept it , I have to use U1RX & U1TX , pins 18 &17, please advise.

vortigont, 2022-10-07:
I do not have any s3 boards in my possession to do tests unfortunately.
Number of pins were limited to original esp32 boards. Need to check if any chip specific macro's exist.

closed 2022-10-07 by vortigont

vortigont, 2022-10-07:
should be fixed with latest commit

commit 090d9ab9 (2022-10-07): make gpio and port num limits in WebUI platform dependent
Closes #26