vortigont/espem #27: ESP32 S2 Mini, No WiFi

Alaa-Aim, 2022-11-06:
@vortigont After compiling for ESP3S2, ESP32-S2FN4R2, the compile & upload everything successful but no AP to configure & Connect to WiFi , below screenshot for board config in *.ini file, Please advise 

[image]

vortigont, 2022-11-07:
Hi @Alaa-Aim 
for this to check I need debug output. Pls check `espem_debug` env in the .ini file. Is that a blank new board that you used?

Alaa-Aim, 2022-11-07:
Hi @vortigont, I've tried this method, but nothing on the serial monitor, this board uses native use not UART, debug section in the *.ini file as below, 
[image]
the board is brand new and blank, I check it by using it as WiFi AP and it works.

Alaa-Aim, 2022-11-07:
This is the board i use

[image]

vortigont, 2022-11-07:
I do not have such boards without serial to test. It may require enabling CDC serial driver.
I'll try to find how to test it when time permits.
Meanwhile I can suggest a feasible workaround - you can upload filesystem image with config file to force enabling AP mode. Put a `config.json` file into `data` dir with a content like this
```
{"APonly":true}```
 Than access AP and configure via GUI as usual

Alaa-Aim, 2022-11-07:
Hi @vortigont , Unfortunately, didn't work, I'll try use UART, I'll let know the outcome.

Alaa-Aim, 2022-11-08:
> you may also need to specify `board = lolin_s2_mini` [lolin_s2_mini.json](https://github.com/platformio/platform-espressif32/blob/master/boards/lolin_s2_mini.json) [S2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html)

Hi @vortigont , Changing the board to lolin_s2_mini , works , many thanks

closed 2022-11-08 by Alaa-Aim