vortigont/espem #28: Erorr Compling for ESP32S3

Alaa-Aim, 2022-12-27:
@vortigont , after the last update, tried to compile, but i had the error below :

Compiling .pio\build\espem\libf53\pzem-edl\modbus_crc16.cpp.o
Compiling .pio\build\espem\libf53\pzem-edl\msgq.cpp.o
Compiling .pio\build\espem\libf53\pzem-edl\pzem_edl.cpp.o
Compiling .pio\build\espem\libf53\pzem-edl\pzem_modbus.cpp.o
.pio/libdeps/espem/EmbUI/EmbUI/ui.cpp: In member function 'void Interface::json_section_extend(const String&)':
.pio/libdeps/espem/EmbUI/EmbUI/ui.cpp:186:89: error: 'class ArduinoJson6200_F1::JsonArray' has no member named 'getElement'
     json_section_begin(name, (char*)0, false, false, false, section_stack.tail()->block.getElement(section_stack.tail()->block.size()-1));    // find last array element
                                                                                         ^~~~~~~~~~
Compiling .pio\build\espem\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\espem\FrameworkArduino\FirmwareMSC.cpp.o
scons: *** [.pio\build\espem\libbbb\EmbUI\ui.cpp.o] Error 1
==================================================================== [FAILED] Took 40.87 seconds ====================================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:40.870
=============================================================== 1 failed, 0 succeeded in 00:00:40.870, please advise

vortigont, 2022-12-27:
Hi @Alaa-Aim 
looks like there were some breaking changes in the [latest](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.20.0) release of ArduinoJson that affected EmbUI lib.
Let me check this and find a proper way to fix.

Alaa-Aim, 2022-12-27:
@vortigont thank you fro prompt response, waiting for the fix.

vortigont, 2022-12-28:
fixed in [EmbUI](https://github.com/vortigont/EmbUI/commit/4284b6a1d0ff55d6241c8ea2903def4ef995117f)
might need updating caches/resource files on littlefs

closed 2022-12-28 by vortigont

Alaa-Aim, 2022-12-28:
Hi @vortigont , unfortunately, I still getting the same error, even though I did clean file & compiling, please advise.

Compiling .pio\build\espem\libbbb\EmbUI\ui.cpp.o
Compiling .pio\build\espem\libf53\pzem-edl\modbus_crc16.cpp.o
Compiling .pio\build\espem\libf53\pzem-edl\msgq.cpp.o
Compiling .pio\build\espem\libf53\pzem-edl\pzem_edl.cpp.o
Compiling .pio\build\espem\libf53\pzem-edl\pzem_modbus.cpp.o
.pio/libdeps/espem/EmbUI/EmbUI/ui.cpp: In member function 'void Interface::json_section_extend(const String&)':
.pio/libdeps/espem/EmbUI/EmbUI/ui.cpp:186:89: error: 'class ArduinoJson6200_F1::JsonArray' has no member named 'getElement'
     json_section_begin(name, (char*)0, false, false, false, section_stack.tail()->block.getElement(section_stack.tail()->block.size()-1));    // find last array element
                                                                                         ^~~~~~~~~~
Compiling .pio\build\espem\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\espem\FrameworkArduino\FirmwareMSC.cpp.o
scons: *** [.pio\build\espem\libbbb\EmbUI\ui.cpp.o] Error 1
==================================================================== [FAILED] Took 44.85 seconds ====================================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:44.845

Alaa-Aim, 2022-12-28:
@vortigont , I had to update the ui.cpp file manually, as PIO couldn't pull the updated file, once ui.cpp updated, everything is fine, thanks

vortigont, 2022-12-28:
ah... I see, I have set strict dependency for espem on 2.7.0 release tag for EmbUI, so that is why is still pulls previous version.
I will change it for master branch for now and rebind to 2.7.1 once I'll release it. It contains some other important fixes also.

Alaa-Aim, 2022-12-28:
Hi @vortigont , web page only Russian, even if select the English, only setting page change to English, others stay Russian, could you please check that , thanks in advanced

vortigont, 2022-12-28:
huh, good catch! you are the best tester, @Alaa-Aim  :)
fixed!

Alaa-Aim, 2022-12-28:
@vortigont many thanks for your efforts and prompt actions.

Alaa-Aim, 2022-12-28:
Hi @vortigont , unfortunately , I have error compiling as below
In file included from espem/interface.cpp:4:
espem/ui_i18n.h:60:3: error: expected '}' before '{' token
   { T_RU_ESPEM,
   ^
espem/ui_i18n.h:50:59: note: to match this '{'
 static const char *const C_DICT[][UI_DICT_SIZE] PROGMEM = {
                                                           ^
espem/ui_i18n.h:60:3: error: expected ',' or ';' before '{' token
   { T_RU_ESPEM,
   ^
espem/ui_i18n.h:66:4: error: expected unqualified-id before ',' token
   },
    ^
espem/ui_i18n.h:67:1: error: expected unqualified-id before '}' token
 };
 ^
espem/ui_i18n.h:67:1: error: expected declaration before '}' token
Compiling .pio\build\espem\lib246\AsyncMqttClient\AsyncMqttClient\Packets\ConnAckPacket.cpp.o
*** [.pio\build\espem\src\interface.cpp.o] Error 1
==================================================================== [FAILED] Took 12.84 seconds ====================================================================
Environment    Status    Duration
-------------  --------  ------------
espem          FAILED    00:00:12.842

vortigont, 2022-12-28:
ah, sorry I missed commas :) try now
strange that CI builder did not get the error, hmmm...

Alaa-Aim, 2022-12-28:
[image]
  I fix it already, many thanks

Alaa-Aim, 2022-12-31:
Hi @vortigont , I've tried to utilize the PSRAM by moving all the heap to the PSRAM ( my board have 4MB PSRAM), I couldn't mange to do that, I've looked & searched but I didn't find the right info in this regard, can you please on this ? really appreciate it and thank you in advanced =

vortigont, 2023-01-01:
Hi @Alaa-Aim, Happy New Year to you too! :)
I'm not sure what do you mean by moving all heap to PSRAM. But this is bad idea in general, PSRAM makes sense to store large and infrequently used data. Otherwise it will only slow down everything else. And not all structures and objects could be placed in PSRAM.

Alaa-Aim, 2023-01-01:
Hi @vortigont , thanks for your response, my idea is to buffer large data to PSRAM, as SRAM is small (around 300KB), I want to buffer the reading of PZEM every second or every minute (buffering like 10 to 20 readings), then send it to the cloud, using the internal RAM, puts a pressure on ESP32 and slow everything down, is my idea feasible ?