vortigont/espem #38: Ошибки при компиляции (линковке)

DmitriFed, 2023-10-09:
Добрый вечер!
При попытке компилирования ошибки:
Building in release mode
Linking .pio\build\espem\firmware.elf
c:/users/dmitri/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\espem\lib702\libEmbUI.a(ftpsrv.cpp.o): in function `ftp_loop()':
C:\Users\Dmitri\Documents\Platformio\Projects\espem-3.1.0/.pio/libdeps/espem/EmbUI/EmbUI/ftpsrv.cpp:20: multiple definition of `ftp_loop()'; .pio\build\espem\src\main.cpp.o:C:\Users\Dmitri\Documents\Platformio\Projects\espem-3.1.0/espem/ftpSrv.h:33: first defined here
collect2.exe: error: ld returned 1 exit status

closed 2023-10-10 by vortigont

vortigont, 2023-10-10:
Добрый! Странно, этот заголовочник нигде не вкладывается, по-идее вообще не должен был учитываться.
В любом случае надо было удалить, спасибо!

vortigont, 2023-10-11:
вы не забыли образ ФС залить? `pio run -t uploadfs`

DmitriFed, 2023-10-13:
Добрый вечер!
Забыл ( Не до конца прочитал "build.md". С esp32 (соответственно, с необходимостью загрузки файловой системы)  раньше дела не имел, только с ардуинами разными. ФС залил, прошивка заработала, датчик увидела,  данные показывает. 
Дальше я пытаюсь настроить скрипты  сохранения данных в базу для отображения графиков на сервере. Собственно, ради чего это всё и делается. Поднял локальный сервер под windows, скачал и подкинул smarty, поправил пути и вместо имени устройства задал его ip в config.php . 
Страница "Энергомер - статистика"  на localhost с меню открывается и все. Самой статистики нет. В лог php сыпятся ошибки, я так понимаю от smarty. Тут я и застрял (( В php, ajax я практически полный нуб.
Был бы очень благодарен, если бы Вы смогли немного подсказать по настройке взаимодействия прошивки и серверной части
[php_error.log](https://github.com/vortigont/espem/files/12895495/php_error.log)

[config.php.txt](https://github.com/vortigont/espem/files/12895487/config.php.txt)

vortigont, 2023-10-15:
Ух... вообще этот кусок с вебом на пхп и смарти довольно древняя вещь, я его забросил. Сейчас модно всякие графаны и прометеусы использовать для графиков, а данные собирать агентами. Но это повозиться надо.
Для того что есть в проекте на смарти нужно еще базу данных завести, куда данные будут собираться и складываться. Можно использовать MySQL или SQLite, схема с базой лежит [тут](https://github.com/vortigont/espem/tree/master/sql). Там по сути 2 таблицы, список датчиков и данные. Параметры и тип базы настраиваются в файле `config.php`. В том конфиге что вы прицепили база не настроенна, соответвествтенно у вас ничего собираться не будет. Для сборки данных нужно через крон наcтроить вызов скрипта `www/inc/pzem_poller.php`, он может писать данные в лог файл и в базу, если настроенно.
Под винду, это все запустить будет весьма незаурядно, я с виндой не дружу :)

Я планирую в скором времени добавить полную поддержку mqtt в проект, можно будет собирать данные через mqtt если с ним дружите.

commit b21515dd (2023-10-10): remove obsolete ftp header

closes #38