vortigont/espem #44: Repeated Crashes

Alaa-Aim, 2024-04-28:
@vortigont , Hi, Hope you have a lovely day.
after compile and upload the new update, on ESP32S3, 4MB with no PSRAM, it keeps crashing in both espem and espem_dummy, the log as below:
'Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420c677e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[ 13653][E][vfs_api.cpp:105] open(): /littlefs/generate_204 does not exist, no permits for creation
[ 13656][E][vfs_api.cpp:105] open(): /littlefs/generate_204.gz does not exist, no permits for creation
[ 13663][E][vfs_api.cpp:105] open(): /littlefs/generate_204/index.html does not exist, no permits for creation
[ 13673][E][vfs_api.cpp:105] open(): /littlefs/generate_204/index.html.gz does not exist, no permits for creation
[ 13686][E][vfs_api.cpp:105] open(): /littlefs/index.html does not exist, no permits for creation
[ 15721][E][vfs_api.cpp:105] open(): /littlefs/generate_204.gz does not exist, no permits for creation
[ 15723][E][vfs_api.cpp:105] open(): /littlefs/generate_204 does not exist, no permits for creation
[ 15730][E][vfs_api.cpp:105] open(): /littlefs/generate_204/index.html.gz does not exist, no permits for creation
[ 15740][E][vfs_api.cpp:105] open(): /littlefs/generate_204/index.html does not exist, no permits for creation
[ 16189][E][vfs_api.cpp:105] open(): /littlefs/generate_204.gz does not exist, no permits for creation
[ 16193][E][vfs_api.cpp:105] open(): /littlefs/generate_204 does not exist, no permits for creation
[ 16198][E][vfs_api.cpp:105] open(): /littlefs/generate_204/index.html.gz does not exist, no permits for creation
[ 16207][E][vfs_api.cpp:105] open(): /littlefs/generate_204/index.html does not exist, no permits for creation
[ 16438][E][vfs_api.cpp:105] open(): /littlefs/icon-144x144.png.gz does not exist, no permits for creation
[ 16484][E][vfs_api.cpp:105] open(): /littlefs/favicon.png.gz does not exist, no permits for creation
[ 16530][E][vfs_api.cpp:105] open(): /littlefs/generate_204.gz does not exist, no permits for creation
[ 16533][E][vfs_api.cpp:105] open(): /littlefs/generate_204 does not exist, no permits for creation
[ 16539][E][vfs_api.cpp:105] open(): /littlefs/generate_204/index.html.gz does not exist, no permits for creation
[ 16548][E][vfs_api.cpp:105] open(): /littlefs/generate_204/index.html does not exist, no permits for creation
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420062c7  PS      : 0x00060030  A0      : 0x82007361  A1      : 0x3fcebb00  
A2      : 0x3fcebe20  A3      : 0x3fcebb28  A4      : 0x3fc94d98  A5      : 0x3fc9a9ec
A6      : 0x3fcebb10  A7      : 0x00000001  A8      : 0x820062c2  A9      : 0x3fcebae0  
A10     : 0x00000000  A11     : 0x3c0ea0fe  A12     : 0x3fcebb10  A13     : 0x3fcebe40
A14     : 0x00000000  A15     : 0x3fceba48  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffe  

Backtrace: 0x420062c4:0x3fcebb00 0x4200735e:0x3fcebbe0 0x420c4353:0x3fcebd90 0x42015ded:0x3fcebdb0 0x420165d2:0x3fcebe00 0x4201661a:0x3fcebe90 0x4201e79a:0x3fcebec0 
0x4201e91d:0x3fcebee0 0x42015b86:0x3fcebf00 0x42007c96:0x3fcebf20 0x420225d9:0x3fcebf40

  #0  0x420062c4:0x3fcebb00 in ArduinoJson::V6215PB2::detail::MemberProxy<ArduinoJson::V6215PB2::JsonObject, char const*>::getOrCreateData() const at .pio/libdeps/espem_dummy/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:54
      (inlined by) ArduinoJson::V6215PB2::detail::VariantData* ArduinoJson::V6215PB2::detail::VariantAttorney::getOrCreateData<ArduinoJson::V6215PB2::detail::MemberProxy<ArduinoJson::V6215PB2::JsonObject, char const*> const>(ArduinoJson::V6215PB2::detail::MemberProxy<ArduinoJson::V6215PB2::JsonObject, char const*> const&) at .pio/libdeps/espem_dummy/ArduinoJson/src/ArduinoJson/Variant/VariantAttorney.hpp:44
      (inlined by) ArduinoJson::V6215PB2::detail::VariantRefBase<ArduinoJson::V6215PB2::detail::MemberProxy<ArduinoJson::V6215PB2::JsonObject, char const*> >::getOrCreateData() const at .pio/libdeps/espem_dummy/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:274
      (inlined by) ArduinoJson::V6215PB2::detail::VariantRefBase<ArduinoJson::V6215PB2::detail::MemberProxy<ArduinoJson::V6215PB2::JsonObject, char const*> >::getOrCreateVariant() const at .pio/libdeps/espem_dummy/ArduinoJson/src/ArduinoJson/Variant/VariantImpl.hpp:127
      (inlined by) bool ArduinoJson::V6215PB2::detail::VariantRefBase<ArduinoJson::V6215PB2::detail::MemberProxy<ArduinoJson::V6215PB2::JsonObject, char const*> >::set<unsigned int>(unsigned int const&) const at .pio/libdeps/espem_dummy/ArduinoJson/src/ArduinoJson/Variant/VariantImpl.hpp:140
      (inlined by) ArduinoJson::V6215PB2::detail::MemberProxy<ArduinoJson::V6215PB2::JsonObject, char const*>& ArduinoJson::V6215PB2::detail::MemberProxy<ArduinoJson::V6215PB2::JsonObject, char const*>::operator=<unsigned int>(unsigned int const&) at .pio/libdeps/espem_dummy/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:33 
      (inlined by) ui_frame_mkchart(Interface*) at espem/interface.cpp:82
  #1  0x4200735e:0x3fcebbe0 in ui_page_espem(Interface*, ArduinoJson::V6215PB2::JsonObject const*, char const*) at espem/interface.cpp:148 (discriminator 4)
  #2  0x420c4353:0x3fcebd90 in std::_Function_handler<void (Interface*, ArduinoJson::V6215PB2::JsonObject const*, char const*), void (*)(Interface*, ArduinoJson::V6215PB2::JsonObject const*, char const*)>::_M_invoke(std::_Any_data const&, Interface*&&, ArduinoJson::V6215PB2::JsonObject const*&&, char const*&&) at c:\users\alaa\.platformio\packages\toolchain-xtensa-esp32s3\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:297
  #3  0x42015ded:0x3fcebdb0 in std::function<void (Interface*, ArduinoJson::V6215PB2::JsonObject const*, char const*)>::operator()(Interface*, ArduinoJson::V6215PB2::JsonObject const*, char const*) const at c:\users\alaa\.platformio\packages\toolchain-xtensa-esp32s3\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:687   
      (inlined by) ActionHandler::exec(Interface*, ArduinoJson::V6215PB2::JsonObject*, char const*) at .pio/libdeps/espem_dummy/EmbUI/EmbUI/EmbUI.cpp:400
  #4  0x420165d2:0x3fcebe00 in EmbUI::post(ArduinoJson::V6215PB2::JsonObject const&, bool) at .pio/libdeps/espem_dummy/EmbUI/EmbUI/EmbUI.cpp:223
  #5  0x4201661a:0x3fcebe90 in std::_Function_handler<void (), wsDataHandler(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at .pio/libdeps/espem_dummy/EmbUI/EmbUI/EmbUI.cpp:106
      (inlined by) _M_invoke at c:\users\alaa\.platformio\packages\toolchain-xtensa-esp32s3\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:297
  #6  0x4201e79a:0x3fcebec0 in std::function<void ()>::operator()() const at c:\users\alaa\.platformio\packages\toolchain-xtensa-esp32s3\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:687
  #7  0x4201e91d:0x3fcebee0 in Scheduler::execute() at .pio/libdeps/espem_dummy/TaskScheduler/src/TaskScheduler.h:1431
  #8  0x42015b86:0x3fcebf00 in EmbUI::handle() at .pio/libdeps/espem_dummy/EmbUI/EmbUI/EmbUI.cpp:273
  #9  0x42007c96:0x3fcebf20 in loop() at espem/main.cpp:65
  #10 0x420225d9:0x3fcebf40 in loopTask(void*) at C:/Users/ALAA/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: 5e353e5f7522af6d

Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420c677e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
'

vortigont, 2024-04-28:
Hi!
It work for me, at least for dummy mode on esp32. Do not have spare S3 to test for now.
Have you tried to wipe/re-upload FS? According to the traces it crashed on requesting TimeSeries interval data.
I've added a check that should prevent crash, but it won't show interval on the chart in this case.
Pls, try if this helps.

Alaa-Aim, 2024-04-28:
> Hi! It work for me, at least for dummy mode on esp32. Do not have spare S3 to test for now. Have you tried to wipe/re-upload FS? According to the traces it crashed on requesting TimeSeries interval [...]

unfortunately, still the same issue

vortigont, 2024-04-28:
I need more details, does it crashes when you open the UI in browser?
Have you tried to wipe/reupload FS?

Alaa-Aim, 2024-04-28:
> I need more details, does it crashes when you open the UI in browser? Have you tried to wipe/reupload FS?

I did full flash erase using ESP flash download tool, upload the FW then upload FS, firstly, connect to WIFI, after almost 30 sec. disconnect from WIFI and nothing shows on the serial monitor, hard rest it, then it goes in loops, as shows in the log above

vortigont, 2024-04-29:
it can't bootloop with same message because the error above could come only as a respond to opening WebUI in browser.
Pls, try to close the browser and boot the device. If you see crash report, pls post it here.
If you see crash only after you open WebUI in browser it's another case, pls post it here.

Alaa-Aim, 2024-04-29:
> it can't bootloop with same message because the error above could come only as a respond to opening WebUI in browser. Pls, try to close the browser and boot the device. If you see crash report, pls [...]

Hi @vortigont , Hope you have a lovely day,

I did new PIO installation, used new ESP32 board, compile and upload both FW & FS, it works like a charm, later, I'll test it on ESP32S3.

closed 2024-04-29 by Alaa-Aim