vortigont/espem #46: Build problem with v3.2.0

clive0106, 2024-08-05:
Hi
I am not familiar with platformio :(
When I try and build this version via 'platformio run' I get a bunch of errors like..

In file included from .pio/libdeps/espem/esp32-flashz/src/flashz.cpp:30:
.pio/libdeps/espem/esp32-flashz/src/flashz.hpp:40:10: fatal error: Update.h: No such file or directory
 #include <Update.h>
          ^~~~~~~~~~
compilation terminated.

I was wondering if it was a dependancy issue?

I am running the latest platformio core and seem to have the latest updates

I would love to run this

Clive

vortigont, 2024-08-05:
Hey, @clive0106 
thanks for heads-up. Pls, try now, should be fine.
If any issues, try to purge `.pio` folder.

closed 2024-08-05 by vortigont

clive0106, 2024-08-06:
Thank you so much..  It works great.
An impressive project. :)

vortigont, 2024-08-06:
you are welcome!