vortigont/espem #48: Issue compiling ESPEM with AsyncWebServerResponse

elektron44, 2024-09-03:
Dear Vortigont,

Thanks a lot for sharing this project.

I was unfortunately unable to compile this project in PlatformIO, due to a compilation error on line 193 in espem.cpp file: 
AsyncWebServerResponse* response = request->beginChunkedResponse(FPSTR(PGmimejson),
                                  [this, iter, ts](uint8_t* buffer, size_t buffsize, size_t index) mutable -> size_t ....

The error rising is : 
"espem.cpp:193: undefined reference to `AsyncWebServerRequest::beginChunkedResponse(String const&, std::function<unsigned int (unsigned char*, unsigned int, unsigned int)>, std::function<String (String const&)>)' "

When I comment those lines, from 193 to 240, I'm able to compile.

I tried but code is too tricky for me to understand what is going on. 
Can you help me, please?

vortigont, 2024-09-04:
Hey, @elektron44 
pls try the latest commit #4e726c1
should be fine.

elektron44, 2024-09-05:
Thanks a lot @vortigont , It works now.

Well done 👌👍

closed 2024-09-05 by elektron44